MicrosoftDocs/windows-iotcore-docs #132: Software TPM (sTPM) not working after fall creators update

mpoeckl, 2017-11-10:
After updating to Version 10.0.16299.19 the Software TPM (sTMP) is not working any more. 
1. After the update the previous settings where gone on all slots
2. New settings through the Device Portal where stored
3. TPM Settings are are not readable through the actual C# SDKs using the Microsoft.Devices.Tpm Class. All NuGet Packages on the most current versions (Microsoft.NETCore.UniversalWindowsPlatform - 6.0.1, Microsoft.Devices.Tpm - 1.0.0). There is no error but the Device ID shows a random GUID and all other settings are empty Strings
4. Resetting the TPM settings and using different sTPM Slots did not solve the issue

saraclay, 2017-11-21:
Thank you for reaching out. We've filed a bug for this.

ooeygui, 2017-11-22:
The Software TPM is a developer only feature, and work was not done to preserve state on upgrade. Have you attempted to reset the sTPM via the instructions located [here](https://docs.microsoft.com/en-us/windows/iot-core/secure-your-device/setuptpm)?

mpoeckl, 2017-12-01:
Yes I reseted the sTPM multiple times... that the sTPM does not preserve the state during upgrade is not a big problem, however that is not useable at all after the Windows Upgrade is. In the meanwhile this issue is filed as a bug in Windows 10 IoT Core so I hope the issue is resolved in the foreseeable future.

closed 2018-01-17 by saraclay