MicrosoftDocs/windows-iotcore-docs #153: Community provided article on coding for Windows devices

andygonusa, 2017-12-13:
Hi all, attached please find an article submitted by a Brazilian community MVP. Originally it was published on a local section of the MSDN library and had very good PVs. It provides a good overall description of coding for devices / iot.

It has been reverse-machine-translated to provide a readable English version - if you think it has some merit and could find a place in the official documentation, please do let me know and we can work with the author to have it amended as necessary.

This also serves as a test, to see if it's efficient to use this process for such submissions/proposal.

thanks to let me know your thoughts
[Trabalhando com sensores em UWP_Parte 1_Sensor de Luz.zip](https://github.com/MicrosoftDocs/windows-iotcore-docs/files/1555600/Trabalhando.com.sensores.em.UWP_Parte.1_Sensor.de.Luz.zip)

saraclay, 2018-01-10:
Hello Andy, thank you for submitting this article. Let me ask the team how we can make the best use of it!

saraclay, 2018-01-10:
Hello again Andy,

I wanted to thank you again for submitting this article. Typically, MVPs will post this content on their own blog. Have you already posted this on your blog?

andygonusa, 2018-01-16:
Hi Sara,
We are working with Jennifer on the possibilities of posting customer articles into Docs. Please ignore for now.

Thanks,
Andy

closed 2018-01-17 by saraclay