MicrosoftDocs/windows-iotcore-docs #162: Os couldn't boot up

ramtinak, 2018-01-09:
Hi, I'm using Rspi 3 model B, and windows for iot build 16299 or previous version. 
I find out a bug in all Win IOTs! When you unplug rspi from power (accidentally or not) a couple times(or sometimes one time) Windows couldn't boot up and you have to install windows again!
In this scenario when you trying to power on the Rspi , just shows you an blank black page with anything else, even windows logo couldn't show up!

What is the problem?

saraclay, 2018-01-10:
Hello,

Are you sure that you've flashed your devices according to our documentation using the IoT Dashboard?

ramtinak, 2018-01-10:
Yes, I'm pretty sure. 

My friend has same problem, as I said before,  When you unplug rspi from power supplies(cable)(accidentally or not) sometimes Os couldn't boot up, and you most install Windows.

P.s: When you shutdown the Os this problem never happened but when power goes off or Os is not currently shutdown this issue appear.

saraclay, 2018-01-10:
This may be a problem with the Raspberry Pi 3 itself. Please see this forum: https://www.raspberrypi.org/forums/viewtopic.php?t=59652

closed 2018-01-17 by saraclay