MicrosoftDocs/windows-iotcore-docs #168: HardwareCompatList doesn't mention the DragonBoard 410c at all

naikrovek, 2018-01-18:
None of the officially compatible hardware for Windows 10 IoT Core shows compatibility with the DragonBoard 410c, even though I know personally that many of these devices work fine.

I suspect that MS has done the testing on these devices, and simply hasn't updated their documentation to reflect that.

Please update the documentation to reflect the status of compatibility with the only supported development platform not listed - the DragonBoard 410c.

saraclay, 2018-01-31:
Hello @naikrovek -

Have you gone through our Get Started flow? DragonbBoard 410c is definitely supported. I'm assuming the fragmentation of the information may be confusing... I will look into how this can be improved.

https://developer.microsoft.com/en-us/windows/iot/getstarted/prototype/selectdevice

naikrovek, 2018-01-31:
Yeah I know the board is supported, that's not the issue.  It's that the DragonBoard isn't listed as compatible with certain bits of hardware listed on the supported peripherals page.  Only the RPi2, RPi3, and MBM are mentioned.

This page: https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/1349cea6697410f39dd8ca7ed43cb3cf00de042d/windows-iotcore/learn-about-hardware/HardwareCompatList.md

saraclay, 2018-02-09:
Got it. We've made changes to the [Hardware CompatibilityList](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/hardwarecompatlist ) thanks to your suggestion.

closed 2018-02-14 by saraclay