MicrosoftDocs/windows-iotcore-docs #249: Inaccurate Information on Step 3: Download and Install

scout208, 2018-03-13:
The instructions under Step 3: Download and Install seem to be incorrect. There is no option available for **Download Windows 10 package** under **App Management** > **Current Packages** like the instructions say there is.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 23c23312-2bd2-e489-eedc-41afd354584c
* Version Independent ID: 60ca62fb-2538-759e-e7b5-a89b2ab6c41c
* Content: [Microsoft Store support for IoTCore Apps - Windows IoT | Microsoft Docs](https://docs.microsoft.com/en-us/windows/iot-core/commercialize-your-device/installingandservicing#feedback)
* Content Source: [windows-iotcore/commercialize-your-device/InstallingAndServicing.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/commercialize-your-device/InstallingAndServicing.md)
* Service: **unspecified**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

saraclay, 2018-03-13:
Hello,

It looks like you do not have pre-install permissions. To get those, you need to fill out this form: https://forms.office.com/Pages/ResponsePage.aspx?id=v4j5cvGGr0GRqy180BHbR4oorCHCXCJHla0pHy8o5GpUMlVHVkNZNVVON1ZTVUQ0T0lHS1pDV0JFVy4u

scout208, 2018-03-13:
@saraclay Thanks but I have filled out thatl form before. Should I do it again anyways?

saraclay, 2018-03-13:
Yes - I wonder why it didn't go through last time.

scout208, 2018-03-14:
@saraclay My Github account is associated with a different email than the one I am using for Windows Dev Center. The email for the account for which I need pre-install permissions for is cearly@monarcsport.com. I noticed the form had been updated since the last time I had filled it, so I filled out the form again. I received more of a response than last time as well, so I'm hoping this will solve this issue for me. I have yet to follow the pre-install instructions that were emailed to me, but I will close this issue if they work.

scout208, 2018-03-14:
So it appears that worked. I have yet to test the package that I was able to download, but if I have any trouble with that I think that should be a separate issue. Thanks for helping me get that sorted out @saraclay!

closed 2018-03-14 by scout208

saraclay, 2018-03-14:
Neat - glad to hear it :)

neverthry, 2018-03-18:
Has anybody managed to make it work to update your apps on Win 10 IoT directly from the Windows Store?