MicrosoftDocs/windows-iotcore-docs #255: Unable to Start App Installed from App Package Downloaded in Step 3

scout208, 2018-03-26:
I installed the app on my device using the Device Portal, but I'm unable to start the app. Everytime I try to start it I get a popup window saying "Failed to start {app name}".

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 23c23312-2bd2-e489-eedc-41afd354584c
* Version Independent ID: 60ca62fb-2538-759e-e7b5-a89b2ab6c41c
* Content: [Microsoft Store support for IoTCore Apps - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/commercialize-your-device/InstallingAndServicing#step-3--download-and-install)
* Content Source: [windows-iotcore/commercialize-your-device/InstallingAndServicing.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/commercialize-your-device/InstallingAndServicing.md)
* Service: **unspecified**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

saraclay, 2018-03-29:
Hi @scout208, did your device have access to the Internet when you tried installing and launching the app? It's possible that without giving your device access to the Internet this can throw things off.

saraclay, 2018-03-29:
Got it. Give that a try and let me know!

scout208, 2018-03-30:
You were right, @saraclay. After connecting my device to the internet and installing and starting the app, it started up fine. Maybe it would be good to include that in the documentation [here ](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/appinstaller#using-windows-device-portal)or something. I will close this issue for now.

closed 2018-03-30 by scout208