MicrosoftDocs/windows-iotcore-docs #262: No "BSPFM.xml" file in "BSPPKG_DIR" folder of the DragonBoard 410c

moyanming, 2018-04-02:
I refer to the step:
https://docs.microsoft.com/zh-cn/windows/iot-core/build-your-image/createbsps#dragonboard-410c
bsp is "a8016aaattnwza21120000.2_db410c_bsp".
I have export the bsp cab files and set the BSPPKG_DIR to the "C:\MyBSPs\QCDB410C" folder successfully, but there is no "BSPFM.xml" file.
But it seems that the "buildpkg all" command runs ok.

The following is the prompt:
IoTCoreShellv4.1 arm 10.0.0.0
C:\IoT-ADK-AddonKit\Tools>set BSPPKG_DIR=c:\MyBSPs\QCDB410C

IoTCoreShellv4.1 arm 10.0.0.0
C:\IoT-ADK-AddonKit\Tools>buildpkg all
Building all provisioning packages
Processing all provisioning packages
 Processing Provisioning.Enroll
 Processing Appx.DigitalSign
 Processing Appx.IoTCoreDefaultApp
 Processing Appx.IoTOnboardingTask
 Processing DigitalSign
 Processing MultiLangSample
 Processing RecoverySample
 Processing SampleA
 Processing SampleB
 Processing SecureSample
 Processing SingleLangSample
Building all packages under C:\IoT-ADK-AddonKit\Common\Packages
 Processing Custom.BCD.wm.xml
 Processing Custom.Cmd.wm.xml
 Processing Custom.OOBEApp.wm.xml
 Processing Custom.Settings.wm.xml
 Processing Device.SystemInformation.wm.xml
 Processing DeviceLayout.GPT4GB.wm.xml
 Processing DeviceLayout.GPT8GB-R.wm.xml
 Processing DeviceLayout.MBR4GB.wm.xml
 Processing DeviceLayout.MBR8GB-R.wm.xml
 Processing Provisioning.Auto.wm.xml
 Processing Provisioning.Enroll.wm.xml
 Processing Recovery.GPT-BCD.wm.xml
 Processing Recovery.GPT-BcdEdit.wm.xml
 Processing Registry.Version.wm.xml
 Processing Security.Bitlocker.wm.xml
 Processing Security.DeviceGuard.wm.xml
 Processing Security.SecureBoot.wm.xml
 Processing Settings.HotKey.wm.xml
Building all packages under C:\IoT-ADK-AddonKit\Source-arm\Packages
 Processing Appx.DigitalSign.wm.xml
 Processing Appx.IoTCoreDefaultApp.wm.xml
 Processing Appx.IoTOnboardingTask.wm.xml
 Processing AzureDM.Services.wm.xml
Building all bsp packages
 Processing CustomRPi2.GPIO.wm.xml
 Processing Custom.SMBIOS.wm.xml
 Processing QCDB410C.DeviceLayout-R.wm.xml
 Processing QCDB410C.DevicePlatform-R.wm.xml
 Processing RPi.Customization.wm.xml
-----build done-----

IoTCoreShellv4.1 arm 10.0.0.0
C:\IoT-ADK-AddonKit\Tools>

---
#### 文档详细信息

⚠ *请勿编辑此部分。 docs.microsoft.com ➟ GitHub 问题链接必须具有此部分。*

* ID: 65525f32-8400-1cbc-6c15-20cb6b9a3ca1
* Version Independent ID: 77aea99a-f2b1-da90-beeb-88fa0b2d0708
* Content: [Creating Board Supported Packages (BSP) - Windows IoT](https://docs.microsoft.com/zh-cn/windows/iot-core/build-your-image/createbsps#dragonboard-410c)
* Content Source: [windows-iotcore/build-your-image/CreateBSPs.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/build-your-image/CreateBSPs.md)
* Service: **unspecified**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

saraclay, 2018-04-02:
What is your question? I don't seem to understand what you're looking for.

parameshbabu, 2018-04-02:
BSPPKG_DIR contains only the prebuilt packages. The BSPFM.xml is part of the iot-adk-addonkit code tree under the bsp\qcdb410c directory - if that's what you are looking for.

moyanming, 2018-04-03:
Hi @parameshbabu, I can't find any file named "BSPFM.xml" in the "iot-adk-addonkit" folder and subdirectory. Where to find this file? By the way, I have successfully build all packages and get the ffu image which is installed successfully on my DragonBoard 410c.

saraclay, 2018-04-03:
@moyanming Paramesh is referring to this directory: https://github.com/ms-iot/iot-adk-addonkit/tree/master/Source-arm/BSP/QCDB410C

moyanming, 2018-04-04:
@saraclay @parameshbabu There is no "BSPFM.xml" file in the directory (https://github.com/ms-iot/iot-adk-addonkit/tree/master/Source-arm/BSP/QCDB410C). But there is a "QCDB410CFM.xml" file in the "Packages" folder. Do you mean the "BSPFM.xml" stands for the "QCDB410CFM.xml" file?

parameshbabu, 2018-04-04:
Yes indeed its that. We use "BSP" as a placeholder to refer to the actual BSP that you use. So in this case it is QCDB410C. Hope this helps.

closed 2018-04-08 by moyanming