MicrosoftDocs/windows-iotcore-docs #264: with Windows 10 IoT

yzzg1025, 2018-04-02:
on connecting

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 563b22e3-1c62-eaad-eb29-6f95b01316ba
* Version Independent ID: 385c19a8-007a-f000-496c-0521606cbf9c
* Content: [Overview of Windows 10 IoT Core - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/windows-iot-core)
* Content Source: [windows-iotcore/windows-iot-core.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/windows-iot-core.md)
* Service: **unspecified**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

yzzg1025, 2018-04-02:
?

saraclay, 2018-04-02:
@yzzg1025 I do not understand what you're asking for. Closing this issue.

closed 2018-04-02 by saraclay