MicrosoftDocs/windows-iotcore-docs #268: Lab 1c: add sample files to the package error

moyanming, 2018-04-08:
DragonBoard 410c with the bsp version "a8016aaattnwza21120000.2_db410c_bsp".

In the step 3 of the section "[Add sample files to the package](https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/add-a-registry-setting-to-an-image#span-idaddsamplefilestothepackagespanadd-sample-files-to-the-package)" (https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/add-a-registry-setting-to-an-image#span-idaddsamplefilestothepackagespanadd-sample-files-to-the-package), after "buildpkg Registry.FilesAndRegKeys" command I got error:
`C:\IoT-ADK-AddonKit\Tools>buildpkg Registry.FilesAndRegKeys

Processing Registry.FilesAndRegKeys.wm.xml

(PkgComposition) : error : SecurityToolBox::CreateCatalog: makecat.exe failed to resign C:\Users\mym\AppData\Local\Temp\z2vxolz2.nk2\update.cat.

Err: 1

Output: opened:    C:\Users\mym\AppData\Local\Temp\4p5kvu20.xyr\THSeika.Registry.FilesAndRegKeys~31bf3856ad364e35~arm~~10.0.0.0

    attribute: OSAttr

    attribute: PackageName

processing: <hash>\\?\C:\Users\mym\AppData\Local\Temp\aptcdkvl.3ad\update.mum\update.mum

processing: <hash>\\?\C:\Users\mym\AppData\Local\Temp\aptcdkvl.3ad\update.mum\arm_thseika-registry-fi..ployment-deployment_31bf3856ad364e35_10.0.0.0_none_c67b4103a72ddd6a.manifest

NOT processed: calculating the indirect data (\\?\C:\Users\mym\AppData\Local\Temp\4jliyynh.n0j\arm_thseika.registry.filesandregkeys_31bf3856ad364e35_10.0.0.0_none_ac33fbc5502becfa\TestFile1.txt)

Failed: CryptCATCDFEnumMembersByCDFTagEx.  Last Error: 0x000003EE

Errors found in parsing the CDF file

Error : Failed to create package. See C:\IoT-ADK-AddonKit\Build\arm\pkgs\logs\Registry.FilesAndRegKeys.wm.log

IoTCoreShellv4.1 arm 10.0.0.0

C:\IoT-ADK-AddonKit\Tools>
`
The "Registry.FilesAndRegKeys.wm.log" is:
`Creating Registry.FilesAndRegKeys.wm.xml Package with version 10.0.0.0 for DigitalSign

Microsoft (C) PkgGen 10.0.10011.16384

--Stack Trace--

   在 Microsoft.Composition.Packaging.PkgManifest.CreateCatalog(String catPath, IEnumerable`1 filesToSign, String packageName, Keyform keyform)

   在 Microsoft.Composition.Packaging.PkgManifest.GenerateAndSignCatalog(Boolean signCatalog)

   在 Microsoft.Composition.Packaging.PkgManifest.SerializeMum(String outputFolder)

   在 Microsoft.Composition.Packaging.PkgManifest.SaveManifest(String outputFolder)

   在 Microsoft.Composition.Packaging.CbsPackage.SavePackage(String outputFolder)

   在 Microsoft.Composition.Packaging.CbsPackage.SaveCab(String cabPath, CompressionType compressionType)

   在 Microsoft.CompPlat.PkgBldr.CabGen.Assembly.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)

   在 Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)

   在 Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)

--End Stack Trace--

Package creation failed with error -1
`
How to fix this problem and add my sample files to the package? I have finished the Lab 1a and 1b successfully.

Best regards.

moyanming, 2018-04-08:
My "Registry.FilesAndRegKeys.wm.xml" file is:

> <?xml version="1.0" encoding="utf-8"?>

<identity xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"

  name="FilesAndRegKeys"

    namespace="Registry"

    owner="$(OEMNAME)"

    legacyName="$(OEMNAME).Registry.FilesAndRegKeys" xmlns="urn:Microsoft.CompPlat/ManifestSchema.v1.00">

    <onecorePackageInfo

        targetPartition="MainOS"

        releaseType="Production"

        ownerType="OEM" />

    <regKeys>

        <regKey

            keyName="$(hklm.software)\$(OEMNAME)\Test">

            <regValue

                name="StringValue"

                type="REG_SZ"

                value="Test string" />

            <regValue

                name="DWordValue"

                type="REG_DWORD"

                value="0x12AB34CD" />

            <regValue

                name="BinaryValue"

                type="REG_BINARY"

                value="12ABCDEF" />

        </regKey>

        <regKey

            keyName="$(hklm.software)\$(OEMNAME)\EmptyKey" />

    </regKeys>

<files>

<file destinationDir="$(runtime.system32)" source="TestFile1.txt" />

<file 

destinationDir="$(runtime.bootDrive)\OEMInstall" source="TestFile2.txt" 

name="filename2.txt" />

</files>

</identity>

saraclay, 2018-04-10:
Make sure that all variable names have been properly replaced.

closed 2018-04-10 by saraclay

moyanming, 2018-04-11:
Hi @saraclay, I will try again this Lab but another person "Connor_Early" have the same proble and he (or she) comment this problem at "Nov 27, 2017":
Can you provide an example of the package definition file, C:\IoT-ADK-AddonKit\Common\Packages\Registry.FilesAndRegKeys\Registry.FilesAndRegKeys.wm.xml, after performing Step 2 under Add sample files to the package. I've tried following the steps, but can't seem to get it right as I keep getting build errors.

moyanming, 2018-04-11:
It seems that there is something wrong. Could you please try this Lab c in a very fresh machine?

WORawSon, 2018-05-17:
Remove the name="TestFile2.txt" example:

<?xml version="1.0" encoding="utf-8"?>
<identity xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
    name="FilesAndRegKeys"
    namespace="Registry"
    owner="$(OEMNAME)"
    legacyName="$(OEMNAME).Registry.FilesAndRegKeys" xmlns="urn:Microsoft.CompPlat/ManifestSchema.v1.00">
    <onecorePackageInfo
        targetPartition="MainOS"
        releaseType="Production"
        ownerType="OEM" />
    <regKeys>
         <regKey
        keyName="$(hklm.software)\$(OEMNAME)\Test">
        <regValue name="StringValue" type="REG_SZ" value="Test string" />
        <regValue name="DWordValue"  type="REG_DWORD" value="0x12AB34CD" />
        <regValue name="BinaryValue" type="REG_BINARY" value="12ABCDEF" />
    </regKey>

        <regKey
            keyName="$(hklm.software)\$(OEMNAME)\EmptyKey" />
    </regKeys>
    <files>
        <file
            destinationDir="$(runtime.system32)"
            source="TestFile1.txt" />
        <file
            destinationDir="$(runtime.bootDrive)\OEMInstall"
            source="TestFile2.txt" />
    </files>
</identity>