MicrosoftDocs/windows-iotcore-docs #274: How to create a headless FFU image?

moyanming, 2018-04-10:
The "Lab 1b" (https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/deploy-your-app-with-a-standard-board) describer how to add an app (headed or headless) to the image. How to create a headless FFU image? Where is the detailed docs about the sample test apps, e.g. IOT_BERTHA, IOT_ALLJOYN_APP? I want know which is the sample test app (I will remove those as the retail need headless) and which is a useful IOT features for headless?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1ab72daf-549b-592f-adaf-9e0e79d6c6ca
* Version Independent ID: 523f5140-e048-0032-a2b5-5d742e2d652e
* Content: [Headed and headless devices - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/headlessmode#feedback)
* Content Source: [windows-iotcore/learn-about-hardware/HeadlessMode.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/HeadlessMode.md)
* Service: **unspecified**
* GitHub Login: @zhuridartem
* Microsoft Alias: **artemz**

saraclay, 2018-04-10:
The feature list can be found here: https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/iot-core-feature-list

closed 2018-04-10 by saraclay

moyanming, 2018-04-11:
Hi @saraclay, I have read all the IOT features. But there is no feature about create the headless FFU image. I want to create a headless FFU image originally but not set a head FFU image to a headless. How to create a headless FFU image? Just remove the "head" APP? What is the default states of the IOT OS, head or headless?