MicrosoftDocs/windows-iotcore-docs #277: Unable to Download Live Kernel Dump

scout208, 2018-04-11:
Clicking **Download live kernel dump** in the **Debug Settings** section of the Device Portal just takes me to a "This page isn't working" page on Google Chrome.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3109c64e-0fff-04f4-d572-84851f558643
* Version Independent ID: 2db48f5c-f8c6-4258-eb6f-1e0710c014f0
* Content: [Windows Device Portal - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/manage-your-device/deviceportal#debugging)
* Content Source: [windows-iotcore/manage-your-device/DevicePortal.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/manage-your-device/DevicePortal.md)
* Service: **unspecified**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-04-18:
Hi @scout208! Taking a look at this right now.

saraclay, 2018-04-23:
We have a bug filed for this. Stay tuned - let us know if this is a blocking issue for you.

saraclay, 2018-05-22:
Apologies @scout208 - we're still waiting to get this fixed. Please let me know if this is a blocker for you - closing this out for now.

closed 2018-05-22 by saraclay