MicrosoftDocs/windows-iotcore-docs #329: remote client on raspberry pi

njctechno, 2018-05-04:
Remote client used to work on the raspberry PI and is very useful. This document is inconsistent as it starts by saying remote client does not work on the raspberry PI, and then finishes to give advice on how to make it work better. Does it or does it not work and if it doesn't can you make the document fully reflect this and can you give any idea when it will be working. Also I am using a forked image to get WIN IOT working on the raspberry PI B+, any idea when the proper image will be available?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e02acee3-6daa-e7b9-f0b8-acd860ae5012
* Version Independent ID: 1eb1c5e5-c86d-b1ab-64b6-bbb24fb5bd04
* Content: [Remote display - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/manage-your-device/remotedisplay)
* Content Source: [windows-iotcore/manage-your-device/RemoteDisplay.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/manage-your-device/RemoteDisplay.md)
* Service: **unspecified**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

cyberh0me, 2018-05-09:
both is correct, currently the Windows IoT Remote client does not work for Raspberry Pi and if it would work the performance on Raspberry Pi is low.

closed 2018-05-10 by saraclay

Bohirjon, 2018-06-19:
@saraclay Hey why is it closed ? is this fixed ?

TorontoFred, 2018-07-28:
This issue is not fixed.   The document is misleading.  Client does not work with Raspberry Pi 3.   However, with Raspbian and VNC.   This site offers three possibilities.   https://eltechs.com/3-ways-to-run-a-remote-desktop-on-raspberry-pi/    This would imply that Remote Client approach is not suitable for Raspberry Pi, not that Raspberry Pi is at fault. The support team at Micro Soft is, in essence, not doing their job considering the importance of this function to IoT.

jordanmills, 2018-08-09:
I just saw that they closed several issues that I reported (unrelated to this).  I think they're under a mandate to keep the number of open issues as close to zero as possible, without actually addressing the issues.