MicrosoftDocs/windows-iotcore-docs #331: Support new boards.

Parsakarami, 2018-05-05:
Is this version supports Raspberry Pi 3 B+ ?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 517239bd-2600-5e4c-b0d7-0df795871df4
* Version Independent ID: 321fa14c-4a84-6010-4eeb-3c793c32988d
* Content: [April 2018 Update - Build 17134 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/commercial/aprilcreatorsupdate)
* Content Source: [windows-iotcore/release-notes/commercial/AprilCreatorsUpdate.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/commercial/AprilCreatorsUpdate.md)
* Service: **unspecified**
* GitHub Login: @DanHarman
* Microsoft Alias: **dan.harman**

cyberh0me, 2018-05-05:
no, currently no ETA

Parsakarami, 2018-05-06:
Does Microsoft work on it or not ? I want to plan on Raspberry Pi3 B+ version and windows Iot for my product.

eskantze, 2018-05-07:
Is Version 17134 currently available on the DB 410?  I  can flash with the FFU 17115, but can't find 17134 - also can't update through windows update as I receive the message "Last Update Failed" continuously.

cyberh0me, 2018-05-07:
@eskantze yes it is, https://www.microsoft.com/en-us/software-download/windows10IoTCore

cyberh0me, 2018-05-09:
There will be a public preview for Raspberry Pi 3 Model B+ soon

closed 2018-05-10 by saraclay