MicrosoftDocs/windows-iotcore-docs #332: Update Process

LNTDS, 2018-05-05:
Hey all,

I am currently attempting the "Windows Update" process on "Windows Device Portal" but IoT Core cannot find the latest April update 17134.

I am running the prior commercial release of IoT, v.10.0.16299.371, Raspberry Pi 3B and have attempted updates today, 18/05/05. Is there a delay until the update is pushed to Windows Update or is the expectation to use the "Windows 10 IoT Core Dashboard" re-flashing the microSD card?

[image]

saraclay, 2018-05-08:
Hi @LNTDS - have you tried again? Sometimes there is a delay.

LNTDS, 2018-05-09:
Hi @saraclay, I just checked and I received the update yesterday evening. Must be a delay for the UK 😛 Thanks!

closed 2018-05-09 by LNTDS