MicrosoftDocs/windows-iotcore-docs #335: Missing document

pauljurczak, 2018-05-07:
There is no __notes for "WiFiDirect supported"__ as mentioned at the beginning of this article.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7e1ef6b3-6d6e-465e-a542-6d6aeddd4fea
* Version Independent ID: c49e4c94-5bfb-43a3-42e6-12e00f299a3d
* Content: [Using WiFi Direct on your Windows 10 Iot Core Device - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/setupwifidirect)
* Content Source: [windows-iotcore/connect-your-device/SetupWifiDirect.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/connect-your-device/SetupWifiDirect.md)
* Service: **unspecified**
* GitHub Login: @derekameer
* Microsoft Alias: **demeer**

saraclay, 2018-05-08:
Thanks for finding that, @pauljurczak - we are looking into this.

saraclay, 2018-05-08:
Hi @pauljurczak - that information is actually available under "WiFi Dongles", but the phrasing is confusing. I will be pushing changes out soon to fix that.

closed 2018-05-08 by saraclay