MicrosoftDocs/windows-iotcore-docs #343: Are network priorities fixed in this new update?

Vincation, 2018-05-09:
In the fall creators update a bug was introduced. This bug is not limited to IoT core, but als appeared on Windows for Desktop (since update 1709, 1703 still works fine). 
The problem is that if you have an active ethernet connection without internet, Windows does not route the internet through a USB connected MBIM modem anymore.
A ping test with forced routing works fine, but any other traffic is not routed through the modem. Disconnect the ethernet connector and it starts working.
It seems to be an MBIM driver specific problem since the combination WiFi and LAN without internet works fine.
We tried several things like: leaving the default gateway blank, changing metric interface and managing network adapter priorities (which I understand is created especially for these situations and should work/has worked).
The problem is more thoroughly described here, but this thread seemed to have died quickly: https://social.technet.microsoft.com/Forums/windows/en-US/e20c2af2-f2c1-419c-a2a3-cd2d07d8e699/network-problem-with-mbim-after-upgrading-windows-10-1709.

Only one LAN Port (OR wifi) without internet and a MBIM modem are enough to reproduce te problem. also on a normal Win 10 1709.
You put a static address on the lan cable. no gateway filled in. So "route print"/"netstat -rn" looks fine.
If the cable is connected, no internet trough the cellular modem.
Without a cellular modem: for example one LAN without internet. wifi with internet -> works fine. It gets broken in windows 1709 if a cellular (mbim) modem is in the game. Tested with a telit 910 v2 & Vodafone K5150 with the microsoft stock driver.

The problem is described here : https://social.msdn.microsoft.com/Forums/en-US/6800f827-45d2-4118-9a10-6a8958208e39/network-adapter-priorities-in-iot-broken-after-creators-fall-update?forum=WindowsIoT&prof=required

I'm asking because all channels where I tried to address the issue have died out. I did found several people that have problem with this issue however so we are not the only one.

saraclay, 2018-05-10:
Hi @Vincation,

This is by design. If you have Ethernet or Wi-Fi connected, access to Cellular is blocked.

We have options if you need to bypass Windows policy and allow connectivity over a costed Cellular connection even if Ethernet is connected. Is this Ethernet connection non-routable (e.g. it has no default gateway / no routes)? If so, you can configure the device with a registry value and Windows will allow Cellular traffic even with Ethernet is connected (but not-routable).
 
HKLM\Software\Microsoft\WcmSvc
IgnoreNonRoutableEthernet (REG_DWORD) set to 1

Note that this is not IoT-specific.

Vincation, 2018-05-11:
Hi @saraclay,

This is very good news. We already heard via a partner that it was by design, but got a different registry key which did not do the trick. We're going to test this right away!
The ethernet connection is indeed non routable. Mostly it gets connected to modbus TCP and other Industrial protocol networks via ethernet.

Are these registry keys documented somewhere? Because we have been searching a while for a solution for this issue and could not find anything. 
In the future me might have other applications which require some non-standard settings and it would be nice to be able to fix this without help.

Thank's for the help!

closed 2018-05-22 by saraclay

saadali0, 2020-10-27:
I am having similar problem but adding the above registry did not help in my case. 

In my case i have IoT device that has cellular and two ethernets. 1 ethernet port is used for an IP camera. 

IoT device is running windows 10 pro. 

Cellular is on first priority then the ethernet for internet and i believe there is no route for the internet on ethernet .

From server, I am using SSH tunnel to rdp on the IoT device. rdp works until the ethernet/camera is not plugged in on the port. 

As soon as i plug in the camera, rdp drops. Though ping still seems to work. 

Any thoughts?

guce, 2021-04-20:
Hi @saraclay,

guce, 2021-04-20:
Hi @saraclay,

Is there a solution for routable ethernet now?

Win10 version 1909, both connect cellular and ethernet.  After adding a static route with gateway for the ethernet, the cellular network will be blocked.

Anyone knows? thx

eskilbred, 2021-04-20:
@guce 

The fix is another registry tweak. this one is located in HKEY_LOCAL_MACHINE\SOFTWARE\Policies\Microsoft\Windows\WcmSvc\Local

Create a dword key of fMinimizeConnections and set it to 0

I setup a bunch of PD laptops that use a vpn service over cellular but also dock to a server in their car. This allows the non routable ethernet to still connect without taking priority of cellular.

Hope this helps!

-Erik

guce, 2021-04-21:
> @guce
> 
> Heres the original fix working on v1809
> HKLM\Software\Microsoft\WcmSvc
> IgnoreNonRoutableEthernet (REG_DWORD) set to 1
> 
> The v1909 fix is another registry tweak. this one is located in HKEY_LOCAL_MACHINE\SOFTWARE\Policies\Microsoft\Windows\WcmSvc\Local
> fMinimizeConnections (REG_DWORD) set to 0
> 
> I setup a bunch of PD laptops that use a VPN service over cellular but also dock to a server in their car. This allows the non routable ethernet to still connect without taking priority of cellular.
> 
> Hope this helps!
> 
> -Erik

@eskilbred 

The "fMinimizeConnections" configuration works for me !!!

Now, I can use both the cellular add ethernet (with static route).

Thanks a lot!

Despair2000, 2021-07-10:
Sorry for resurrecting this thread. I have somehow the problem the other way round. My cellular modem always takes priority over my LAN connection which I don't want. I want the cellular just to be there as backup in case my LAN connection goes offline.
I set the interface metric to prefer the LAN connection but this does not make any difference. Can anybody help?