MicrosoftDocs/windows-iotcore-docs #349: Default App change

Ti034, 2018-05-11:
After the update, there was an added default app named "IOTCoreDefaultApplication". The original "IoTCoreDefaultApp" was no longer starting and my boot was stuck at the red circuit board screen. It was able to get into the device with the web interface and noticed the new default app. I set it to the default app and it is working OK now. Did not find this issue documented, so just an FYI in case someone else runs into this.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 517239bd-2600-5e4c-b0d7-0df795871df4
* Version Independent ID: 321fa14c-4a84-6010-4eeb-3c793c32988d
* Content: [April 2018 Update - Build 17134 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/commercial/aprilcreatorsupdate#known-issues)
* Content Source: [windows-iotcore/release-notes/commercial/AprilCreatorsUpdate.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/commercial/AprilCreatorsUpdate.md)
* Service: **windows-iot**
* GitHub Login: @DanHarman
* Microsoft Alias: **dan.harman**

DanHarman, 2018-05-11:
Hi, you are mentioning the wrong danharman github login.

omarbarlas, 2018-05-21:
I have installed this April 2018 Update (Build 17134) on DragonBoard 410, and i am trying to Enable Remote Desktop from Windows Device Portal, but it is not letting me enable that. not sure if i am missing something, or some issue in this release.