MicrosoftDocs/windows-iotcore-docs #350: Unable to unlock SIM PIN from UWP app

Vincation, 2018-05-11:
A modem working with MBIM needs it's SIM unlocked through the MBIM interface and not through AT commands. Through AT commands it will not accept the PIN.

In Windows IoT core this is not possible it seems from code and our device does not have a physical interface (screen/keyboard) or a user to enter it for that matter since it is an IoT device.
I guess it's a remnant from in the mobile phones.
This is the code which works on a Windows 10 desktop, but not on IoT core.

```
var modem = Windows.Networking.NetworkOperators.MobileBroadbandModem.GetDefault();
            Windows.Networking.NetworkOperators.MobileBroadbandDeviceInformation modemDeviceInfo = modem.DeviceInformation;
            Windows.Networking.NetworkOperators.MobileBroadbandPin simpin = modemDeviceInfo.PinManager.GetPin(Windows.Networking.NetworkOperators.MobileBroadbandPinType.Pin1);
            switch (simpin.LockState)
            {
                case Windows.Networking.NetworkOperators.MobileBroadbandPinLockState.PinRequired:
                    var enterPinresult = await simpin.EnterAsync("5966");
                    if (enterPinresult.IsSuccessful)
                    {
                        //Sim unlocked";
                        
                    }
                    else
                    {
                        //"Attempts remaining: " + enterPinresult.AttemptsRemaining.ToString());
                    }
                    break;
                default:
                    break;
            }
```

We have the right app capabilities ([https://docs.microsoft.com/en-us/windows/uwp/packaging/app-capability-declarations](https://docs.microsoft.com/en-us/windows/uwp/packaging/app-capability-declarations)) and should be enough according to MSFT manual.

```
xmlns:rescap="http://schemas.microsoft.com/appx/manifest/foundation/windows10/restrictedcapabilities" 
 
 
<Capabilities>
    <Capability Name="internetClient" />
    <Capability Name="privateNetworkClientServer" />
    <rescap:Capability Name="cellularDeviceIdentity"/>
    <rescap:Capability Name="cellularDeviceControl"/>
    <rescap:Capability Name="cellularMessaging" />
    <DeviceCapability Name="location" />
    <DeviceCapability Name="wiFiControl" />
  </Capabilities>
```

It is not a driver issue or support issue, but a logic error in the operating system. It is probably an artefact from Windows Phone which is not fixed yet.
We also tested the Vodafone (Huawei) K5150 which is officialy supported with the same result.

saraclay, 2018-05-16:
Hi @Vincation, I may be missing some information, but we usually do encourage folks to try and use an external monitor if at all possible.

Vincation, 2018-05-16:
Hey Sara, 
I understand, but it is an IoT device that works on it's own without monitor or even input devices such as keyboard or mouse. It is a standalone device with an integrated modem in a sealed box with a few sensors. And we would like to get it to work with a PIN code still on the SIM.
Unless this is the only solution of course. Is this possible?

cyberh0me, 2018-05-20:
double posting … https://social.msdn.microsoft.com/Forums/en-US/2614f4e2-21f8-40d9-ab76-4ceda0018d7b/unable-to-unlock-sim-pin-from-uwp-app-on-iot-device?forum=WindowsIoT

Vincation, 2018-05-20:
Yes, I tried several channels. Feedback hub, Github, Technet forum, MSDN forum, ... 
But on none of these I got a usefull response. And I find it hard to believe this is by design on an IoT platform. Since enough IoT devices do not require manual user input. So I hope that there might be a registry key or something we could set to make this possible.

saraclay, 2018-05-22:
Hi Vincation, can you file a bug in the Feedback Hub? Thank you.

closed 2018-05-22 by saraclay