MicrosoftDocs/windows-iotcore-docs #353: Questions about running powershell

JeffMushmo, 2018-05-15:
Trying to execute a powershell script from oemcustomizations.cmd. Haven't been able to get it working. Either the powershell script isn't executing, or we don't have elevated enough permissions to execute the script.

A lot of examples show people using net user Administrator to authenticate as admin. My question is, on a retail image where there isn't an administrator account, how does this all work?

Our end goal is that we need to execute a powershell script on first boot to query/edit registry values in HKLM:\System\CurrentControlSet\Enum\

Please advice on best approach.

Also, your link to Custom.cmd gives a 404 error.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ab5535f5-3b3e-503a-ebee-6c79c2af142c
* Version Independent ID: b571e2a2-3ea8-134f-a7c5-f09e2280280c
* Content: [OS Customizations - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/build-your-image/oscustomizations#feedback)
* Content Source: [windows-iotcore/build-your-image/OSCustomizations.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/build-your-image/OSCustomizations.md)
* Service: **windows-iot**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

saraclay, 2018-05-16:
Hi @JeffMushmo - we are looking into this. Stay tuned.

cyberh0me, 2018-05-20:
@JeffMushmo I dont think the oemcustomizations.cmd runs in PowerShell but command line environment...

cyberh0me, 2018-05-26:
@JeffMushmo OEMCustomizations.cmd is invoked by IoTCore Shell (not PowerShell) on every boot with system privileges\
**reg.exe** is the supported way to access the registry [Windows 10 IoT Core Command Line Utils](https://docs.microsoft.com/en-us/windows/iot-core/manage-your-device/commandlineutils#registry-access)

closed 2018-06-07 by saraclay