MicrosoftDocs/windows-iotcore-docs #372: Not able to Enable Remote Desktop on DragonBoard 410c

omarbarlas, 2018-05-21:
I installed the Windows 10 IOT (v17134) on DragonBoard 410c and connected to WiFi and next i logged in to Windows Device Portal, and Tried to enable Remote Desktop, but it fails, The Checkbox goes ON but it says: Failed to enable Windows IOT Remote Server. 
Not sure if i am missing a step or something is not right?

cyberh0me, 2018-05-26:
cant replicate  that issue on 17134 and later

omarbarlas, 2018-05-27:
Yes it fails to enable it every time, here is the screen shot, the Windows 10 IOT version number i have is v.10.0.17134.1
[image]

It always show message: Failed to enable Windows IoT Remote Server

[image]

cyberh0me, 2018-05-27:
can you check if you can control it from command line ?\
ssh into your DragonBoard and
- To query if it is running:
`schtasks.exe /query /tn \Microsoft\Windows\NanoRDP\Start /fo csv /nh`
- To schedule it for startup:
`schtasks.exe /Create /SC ONSTART /TN \Microsoft\Windows\NanoRDP\Start /TR %SystemRoot%\\System32\\NanoRDPServer.exe /RU DefaultAccount /F`
- To Run it:
`schtasks.exe /Run /TN \Microsoft\Windows\NanoRDP\Start`

if this works for you ?

omarbarlas, 2018-05-30:
Ok these commands worked and Remote Desktop Enabled, thanks, Next i am trying to install latest Insider Build Image to make sure the Serial Port issue is resolved

closed 2018-05-30 by omarbarlas

omarbarlas, 2018-07-23:
schtasks.exe /Create /SC ONSTART /TN \Microsoft\Windows\NanoRDP\Start /TR %SystemRoot%\\System32\\NanoRDPServer.exe /RU DefaultAccount /F 
Worked on older release of Windows 10 IOT by running that command directly on the Application Command App in side Windows 10 iot, but what i am trying to Run same on later Insider Build it says Access Denied , Windows 10 IOT Version is:  10.0.17692.1000
[image]