MicrosoftDocs/windows-iotcore-docs #373: SerialDevice.FromIdAsync(entry.Id); always returns null on DragonBoard 410

omarbarlas, 2018-05-21:
I write a Serial Communication UWP App and Deployed same app on DragonBoard 410C and Raspberry Pi3 
It works fine and gets the serial port on Raspberri Pi3 but when i run same code on DragonBoard it returns null as i try to connect to Serial Port in my code:

serialPort = await SerialDevice.FromIdAsync(entry.Id); // This always returns null while entry.ID is correct

saraclay, 2018-05-22:
Hi - can you provide context regarding the documentation you are referring to?

cyberh0me, 2018-05-26:
that's a known issue with Qualcomm DragonBoard 410c BSP Version 2118.0.0.0

[msdn thread](https://social.msdn.microsoft.com/Forums/en-US/0f936077-00ea-4025-b95c-29b60058a8af/17661-dragonboard-410c-no-uart?forum=WindowsIoT)

closed 2018-06-07 by saraclay