MicrosoftDocs/windows-iotcore-docs #374: The c:\EFIESP\EFI\Microsoft\Boot\BCD path is not found

wind-rider, 2018-05-21:
When calling `bcdedit /store c:\EFIESP\EFI\Microsoft\Boot\BCD -dbgsettings serial`, the path is not found. Remote Powershell on the Raspberry PI returns

```
The boot configuration data store could not be opened.
The system cannot find the file specified.
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 59762ca2-1b1d-c2da-dd42-09fbaa0bec47
* Version Independent ID: 0afaadd5-a719-60ae-4b33-242bd9a5fe48
* Content: [Windows Debugger - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/manage-your-device/windowsdebugger)
* Content Source: [windows-iotcore/manage-your-device/WindowsDebugger.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/manage-your-device/WindowsDebugger.md)
* Service: **unspecified**
* Product: **windows-iot**
* GitHub Login: @bfjelds
* Microsoft Alias: **Brian.Fjeldstad**

bfjelds, 2018-05-22:
The EFIESP junction is no longer created.  You'll have to mount it yourself if you want it...you can use mountvol to get the GUID, but I think this is correct for RPi

mkdir C:\EFIESP
mountvol C:\EFIESP \\?\Volume{ae420040-0000-0000-0000-200000000000}

closed 2018-05-22 by saraclay

wind-rider, 2018-05-23:
Could you please update the docs accordingly?

Closed #374<https://github.com/MicrosoftDocs/windows-iotcore-docs/issues/374>.