MicrosoftDocs/windows-iotcore-docs #375: The serial settings for Raspberry PI are unclear

wind-rider, 2018-05-21:
"Fill out the rest of the parameters as appropriate for the Raspberry Pi." is not very helpful.
- Should we use `115200` for the baud settings, or `921600`? Or doesn't it matter?
- What is the com port on the target device (I know how to find the com port for the host device)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5302a6a5-9017-4475-25ee-167a9b816d31
* Version Independent ID: f0bff1be-a4fe-3de0-1ea9-7ec77def3b0c
* Content: [Driver Deployment - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/driverdeployment#step-2--provision-device-with-visual-studio)
* Content Source: [windows-iotcore/learn-about-hardware/DriverDeployment.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/DriverDeployment.md)
* Service: **unspecified**
* Product: **windows-iot**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

saraclay, 2018-05-22:
@wind-rider we are looking into this.

saraclay, 2018-06-07:
Following up here - @wind-rider do you still need help with this?

closed 2018-06-13 by saraclay

wind-rider, 2018-06-13:
@saraclay - sorry for the late reaction. It would still be nice if the documentation would be made more explicit, yes.