MicrosoftDocs/windows-iotcore-docs #382: multi touch error

sajith1988, 2018-05-24:
windows iot last update 17134 error . i used by USB touch frame on my project  that new update error is when we are multi touch on touch frame system is restart .sum time is not restart .

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e550244c-f4be-049b-9131-ba32bd3e9214
* Version Independent ID: 77ea5cc8-8a98-53e4-affd-a1f30538bee6
* Content: [April 2018 Update - Build 17134 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/currentcommercial)
* Content Source: [windows-iotcore/release-notes/CurrentCommercial.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/CurrentCommercial.md)
* Service: **unspecified**
* Product: **windows-iot**
* GitHub Login: @DanHarman
* Microsoft Alias: **dan.harman**

DanHarman, 2018-05-25:
Ok this is getting irritating. Whatever template you use in this repo has my login tagged when I have nothing to do with it. I was told I had been resolved, so can someone here please fix.

cyberh0me, 2018-05-26:
@sajith1988 seems like your devices is not fully USB HID compatible. I would recommend to contact the manufacturer of your touch frame and confirm this device indeed passes the hardware certification tests for Windows 10 HLK / HCK

sajith1988, 2018-06-05:
that before update 10.0.16299.15 is working good .

closed 2018-06-07 by saraclay

cyberh0me, 2018-06-07:
that does not mean its fully USB HID compatible and passes certification tests for Windows 10 HLK / HCK