MicrosoftDocs/windows-iotcore-docs #383: wrong link destination

cyberh0me, 2018-05-25:
IoT Startup App sample link points to https://github.com/Microsoft/Windows-iotcore-samples/tree/develop/Samples/IoTStartApp (a Cpp sample) this link seems ok

IoT Home App sample link points to https://github.com/Microsoft/Windows-iotcore-samples/tree/develop/Samples/IoTHomeAppSample this link seems ok 
but now click on C# and you find an unrelated “IoT Core Media Player” read me ?
And if you click on CS you will find 2 projects IoTHomeApp and IoTStartApp but both projects are C++ not C# so CS is wrong (should be Cpp) and
Additionally compare the “app link path” from before, why do we find 2 projects ? and with wrong path ?
https://github.com/Microsoft/Windows-iotcore-samples/tree/develop/Samples/IoTStartApp/Cpp
vs
https://github.com/Microsoft/Windows-iotcore-samples/tree/develop/Samples/IoTHomeAppSample/CS/IoTStartApp

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9a5be0b6-6fe2-6c0f-79d2-1ae2c3a8dd07
* Version Independent ID: d7a797ed-2dae-ecac-31ba-9c713a10b28a
* Content: [IoT Shell Overview - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/iotcoreshell#switching-between-apps-with-a-home-app)
* Content Source: [windows-iotcore/develop-your-app/IoTCoreShell.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/IoTCoreShell.md)
* Service: **unspecified**
* Product: **windows-iot**
* GitHub Login: @derekameer
* Microsoft Alias: **demeer**

cyberh0me, 2018-06-13:
no, not solved, "IoT Home App" sample link still points to wrong destination 
@saraclay ?

saraclay, 2018-06-20:
This has been resolved.

closed 2018-06-20 by saraclay