MicrosoftDocs/windows-iotcore-docs #390: Additional Resources link broken

makiroggers, 2018-05-29:
The current link to "Cloud and Security Whitepaper" (https://az835927.vo.msecnd.net/sites/iot/Resources/documents/Windows10IoTCoreAzureConnectivityandSecurity.pdf) is broken. When I try to load the URL, it says "The resource you are looking for has been removed, had its name changed, or is temporarily unavailable."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9453a503-3a3e-48aa-7028-5613b531e877
* Version Independent ID: 5d67ee74-3cd8-191a-b696-514443db20b1
* Content: [Overview of IoT on the Cloud - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/connect-to-cloud/overview)
* Content Source: [windows-iotcore/connect-to-cloud/Overview.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/connect-to-cloud/Overview.md)
* Service: **unspecified**
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-06-07:
Thanks for the heads-up, @makiroggers! We'll look into this.

saraclay, 2018-06-08:
The link was a bit dated so we removed it but looking to have an updated whitepaper in time.

closed 2018-06-08 by saraclay