MicrosoftDocs/windows-iotcore-docs #391: You Suck At Web Design

tony-long, 2018-05-29:
To get started with Windows 10 IoT Core, we've created a Windows 10 IoT Core Quickstarter to help you get familiar with the platform quickly.

That takes me to this useless page.   If this was a paper page I'd give it to my puppy to poop on.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 257dc32b-f302-db2a-2571-8f217007d893
* Version Independent ID: 92b2f238-f621-3feb-5b5b-07aed095b115
* Content: [Suggested Prototype Boards - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/tutorials/quickstarter/prototypeboards)
* Content Source: [windows-iotcore/tutorials/quickstarter/PrototypeBoards.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/tutorials/quickstarter/PrototypeBoards.md)
* Service: **unspecified**
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

cyberh0me, 2018-05-30:
@tony-long such disrespectful, unqualified postings without useful and helpful criticism like yours dont help to make things better, they only damage you and your reputation and are waste of time
You should read [GitHub Site Policy](https://help.github.com/categories/site-policy/) especially Respect Others and Be Nice

closed 2018-06-07 by saraclay

Nerobyrne, 2018-06-12:
Actually, if you have ever set up a Raspberry Pi even once, you immediately understand how this works. Not only that, but the links explain very well how to set up Windows 10 IoT on the devices listed.

Also, please keep in mind that this is a page for *developers*. As in, people who know what they are doing before they come here. You're not going to be setting up an IoT dev environment yourself if you don't even know what a "Flash Update Image" is.