MicrosoftDocs/windows-iotcore-docs #401: Can't find info on SD cards compatible with Pi 3

dirtybirdnj, 2018-06-05:
I got windows 10 installed on a microSD card, but for some reason wifi doesn't work. I've read that there are "approved" cards that can be used... but can't find any on the site.

saraclay, 2018-06-07:
Have you tried the SanDisk 64GB microSD card?

cyberh0me, 2018-06-07:
i dont understand the relation between compatible SD card and WiFi doesn't work, please elaborate
[Hardware compatibility list](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/hardwarecompatlist#storage-media) lists supported SD Cards

saraclay, 2018-06-07:
Apologies - I misread the question from @dirtybirdnj. I'm assuming you're using a Raspberry Pi? What kind of RPi are you using?

closed 2018-06-13 by saraclay