MicrosoftDocs/windows-iotcore-docs #403: Download link for Build version 16299

ninhvuonglinh-linhnv, 2018-06-06:
Due to reason Build #17134 is released, from the dashboard I can't choose build #16299 anymore. I need link for download this build. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e87427f3-aa5d-1a31-abf4-3c000cbfe70d
* Version Independent ID: 415bd0f0-8aca-e0a6-11b7-403d34fc2003
* Content: [Fall Creators Update - Build 16299 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/commercial/fallcreatorsupdate)
* Content Source: [windows-iotcore/release-notes/commercial/FallCreatorsUpdate.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/commercial/FallCreatorsUpdate.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-06-07:
Hi @ninhvuonglinh-linhnv, we will make this build available soon on our Downloads page. Stay tuned.

closed 2018-06-07 by saraclay

ninhvuonglinh-linhnv, 2018-06-08:
Hi @saraclay, I found the link:
http://download.microsoft.com/download/9/6/2/9629C69B-02B8-4A82-A4C8-860D6E880C66/16299.15.170928-1534.rs3_release_amd64fre_IOTCORE_RPi.iso
Does it valid link for version 16299 ??