MicrosoftDocs/windows-iotcore-docs #407: Provisioning Package is not Installing

scout208, 2018-06-06:
Hi, I've followed these directions (https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/appinstaller#using-provisioning-package-from-wcd) step-by-step however my app isn't installed even after rebooting.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 51a5db35-a80e-6db9-9648-cfcaabbe19ca
* Version Independent ID: 52a25746-f8eb-fcf3-cc5a-df5533e08451
* Content: [Install your app on an IoT Core device - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/appinstaller#using-provisioning-package-from-wcd)
* Content Source: [windows-iotcore/develop-your-app/AppInstaller.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/AppInstaller.md)
* Product: **windows-iot**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

scout208, 2018-06-06:
My mistake. It must have just taken some time for the app to be installed in the background. It's on there now.

closed 2018-06-06 by scout208

moyanming, 2018-06-08:
Hi @scout208 , I have the same problem. But there is no app was installed after wait about 10 minutes.
I can install the APPX using Windows Device Portal successfully. But the provisioning package .ppkg can't be installed at all.
Any ideas? 
Best regards.

The provisioning package name is "Project_1_1.0.4.0.ppkg", the background application name is "BackgroundApplication1-uwp_s467b7knvf8fe".

[image]

[image]

moyanming, 2018-06-08:
BTW, I uninstall the "ZWaveHeadlessAdapterApp" using the Windows Device Portal, but this app was not reinstall after reboot also. As you can see that the "ZWaveHeadlessAdapterApp.ppkg" is a default .ppkg at the "C$\Windows\Provisioning\Packages" folder.
What is the strategy of the provisioning package?

scout208, 2018-06-08:
@moyanming Like I said, I never really had a problem I just needed to wait a minute for my package to be installed. I wasn't using any background app though, so maybe you could try it with a headed app instead? It might just be a different set of steps for headless apps. 

I don't know anything about ZWaveHeadlessAdapterApp. I just left it as it is.