MicrosoftDocs/windows-iotcore-docs #409: What and where is the "Windows Dev Center"?

moyanming, 2018-06-07:
I want find the Package Family Name (PFN), but What is and where is the "Windows Dev Center"?
--------------
Note
The Package Family Name (PFN) can be found in the Windows Dev Center under App Management > App Identity

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 51a5db35-a80e-6db9-9648-cfcaabbe19ca
* Version Independent ID: 52a25746-f8eb-fcf3-cc5a-df5533e08451
* Content: [Install your app on an IoT Core device - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/appinstaller)
* Content Source: [windows-iotcore/develop-your-app/AppInstaller.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/AppInstaller.md)
* Product: **windows-iot**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

parameshbabu, 2018-06-07:
https://developer.microsoft.com/en-us/windows 
This is the place where you publish your Apps for Store.

closed 2018-06-07 by saraclay

moyanming, 2018-06-08:
Hi @parameshbabu , is there a way to find the PFN that I just test the App but not publish to the Store yet?

parameshbabu, 2018-06-08:
You can use the GetAppxInfo.exe or just use newappxpkg.cmd described in https://aka.ms/iotcoreguide

Regards,
Paramesh