MicrosoftDocs/windows-iotcore-docs #418: Headless App - Target Version

AlexanderPilhar, 2018-06-11:
I recently worked on a updates for two of our Headless Apps (both BackgroundTask running on Windows IoT), also updating Target Version to 17134 and Min. Version to 16299. When trying to run the apps on a remote raspberry pi I encountered errors (entry point not found, app aborting startup, ...). It took me almost two days just to find out I also had to update Target Version and Min Version of the Blank UWP App (which is the app head for my BackgroundTask apps).
Can someone confirm that this is a necessity, please? And if so, put that information in the documentation?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 23c23312-2bd2-e489-eedc-41afd354584c
* Version Independent ID: 60ca62fb-2538-759e-e7b5-a89b2ab6c41c
* Content: [Microsoft Store support for IoTCore Apps - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/commercialize-your-device/installingandservicing)
* Content Source: [windows-iotcore/commercialize-your-device/InstallingAndServicing.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/commercialize-your-device/InstallingAndServicing.md)
* Product: **windows-iot**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

paulmon, 2018-06-11:
The Target Version and Min Version for the Blank App and Headless App need to match.  I will update the documentation

saraclay, 2018-06-12:
The documentation has now been updated.

closed 2018-06-12 by saraclay