MicrosoftDocs/windows-iotcore-docs #444: Error processing request

Colink02, 2018-06-15:
Im not sure why but i want to get the raspberry pi's i tried the Download FFU but when i click confirm on the last one asking for what device it just says error processing request if you could help that be great :)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 257dc32b-f302-db2a-2571-8f217007d893
* Version Independent ID: 92b2f238-f621-3feb-5b5b-07aed095b115
* Content: [Suggested Prototype Boards - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/tutorials/quickstarter/prototypeboards)
* Content Source: [windows-iotcore/tutorials/quickstarter/PrototypeBoards.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/tutorials/quickstarter/PrototypeBoards.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

cyberh0me, 2018-06-18:
@Colink02 please download the image from [Windows 10 IoT Downloads](https://developer.microsoft.com/en-us/windows/iot/Downloads)

@saraclay the download FFU links are pointing to Insider and not to Release, that's not a good idea...

cyberh0me, 2018-06-19:
you are welcome