MicrosoftDocs/windows-iotcore-docs #446: why no download more?

holygoth, 2018-06-17:
Why no download more???

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e816c04b-7c8c-dc54-6382-7c7f2ba532c0
* Version Independent ID: 0e949478-2401-c768-49c6-b21d01d80aa3
* Content: [Get Started Overview - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/tutorials/tutorials)
* Content Source: [windows-iotcore/tutorials/Tutorials.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/tutorials/Tutorials.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

cyberh0me, 2018-06-18:
@holygoth can you clarify the actual issue here as the problem is not clear?

saraclay, 2018-06-27:
Closing this due to ambiguity.

closed 2018-06-27 by saraclay