MicrosoftDocs/windows-iotcore-docs #469: No package definition files found

bessobesso, 2018-06-30:
I download  rpibsp, tried to build but I get error message at the end: Error: No package definition files found. What am I doing wrong?
c:\rpibsp&gt;build.cmd
signing c:\rpibsp\bcm2836pwm.sys
signing c:\rpibsp\bcm2836sdhc.sys
signing c:\rpibsp\bcmauxspi.sys
signing c:\rpibsp\bcmgpio.sys
signing c:\rpibsp\bcmi2c.sys
signing c:\rpibsp\bcmspi.sys
signing c:\rpibsp\pi_miniuart.sys
signing c:\rpibsp\rpiq.sys
signing c:\rpibsp\rpisdhc.sys
signing c:\rpibsp\rpiwav.sys
signing c:\rpibsp\SerPL011.sys
signing c:\rpibsp\vchiq.sys
signing c:\rpibsp\vchiq_arm_kern.dll
signing c:\rpibsp\vcos_win32_kern.dll
Building packages in c:\rpibsp\
Error: No package definition files found.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 65525f32-8400-1cbc-6c15-20cb6b9a3ca1
* Version Independent ID: 77aea99a-f2b1-da90-beeb-88fa0b2d0708
* Content: [Creating Board Supported Packages (BSP) - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/build-your-image/createbsps)
* Content Source: [windows-iotcore/build-your-image/CreateBSPs.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/build-your-image/CreateBSPs.md)
* Product: **windows-iot**
* GitHub Login: @parameshbabu
* Microsoft Alias: **pabab**

parameshbabu, 2018-06-30:
Use this link for the rpibsp : https://github.com/ms-iot/iot-adk-addonkit/releases/download/v4.4/rpibsp-wm.zip. I have updated the docs as well.

bessobesso, 2018-06-30:
OK, I did it, but now I get "Failed DoConvert" error

c:\rpibsp>build.cmd
signing c:\rpibsp\bcm2836pwm.sys
signing c:\rpibsp\bcm2836sdhc.sys
signing c:\rpibsp\bcmauxspi.sys
signing c:\rpibsp\bcmgpio.sys
signing c:\rpibsp\bcmi2c.sys
signing c:\rpibsp\bcmspi.sys
signing c:\rpibsp\pi_miniuart.sys
signing c:\rpibsp\rpiq.sys
signing c:\rpibsp\rpisdhc.sys
signing c:\rpibsp\rpiwav.sys
signing c:\rpibsp\SerPL011.sys
signing c:\rpibsp\vchiq.sys
signing c:\rpibsp\vchiq_arm_kern.dll
signing c:\rpibsp\vcos_win32_kern.dll
Building packages in c:\rpibsp\
 Processing bcm2836pwm.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\bcm2836pwm.wm.log
 Processing bcm2836sdhc.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\bcm2836sdhc.wm.log
 Processing bcmauxspi.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\bcmauxspi.wm.log
 Processing bcmgpio.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\bcmgpio.wm.log
 Processing bcmi2c.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\bcmi2c.wm.log
 Processing bcmspi.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\bcmspi.wm.log
 Processing piminiuart.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\piminiuart.wm.log
 Processing rpiq.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\rpiq.wm.log
 Processing rpisdhc.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\rpisdhc.wm.log
 Processing rpiwav.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\rpiwav.wm.log
 Processing SerPL011.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\SerPL011.wm.log
 Processing SV.PlatExtensions.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\SV.PlatExtensions.wm.log
 Processing vchiq.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\arm\pkgs\logs\vchiq.wm.log

parameshbabu, 2018-07-02:
I suppose you have an non-English install.

The Fix for this is :
 "Settings" -> "Time and Language" -> "Region and Language" -> "Administrative Language Settings" -> "Copy settings..." -> Select "Welcome screen and system accounts" -> click "Ok" to reboot system.
Can you try this? https://github.com/ms-iot/iot-adk-addonkit/issues/239 
If you still have issues, please follow up on the tools GitHub, thanks.

bessobesso, 2018-07-02:
Thank you very much. That worked.

saraclay, 2018-07-03:
Thanks @parameshbabu!

closed 2018-07-03 by saraclay