MicrosoftDocs/windows-iotcore-docs #470: Default port 8080

bessobesso, 2018-07-01:
Changing of default port does not work for me. I am able to write and read the registry values, but webmanagment dont care about this, it responds only if the port is set to 8080; What am I doing wrong?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3109c64e-0fff-04f4-d572-84851f558643
* Version Independent ID: 2db48f5c-f8c6-4258-eb6f-1e0710c014f0
* Content: [Windows Device Portal - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/manage-your-device/deviceportal#feedback)
* Content Source: [windows-iotcore/manage-your-device/DevicePortal.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/manage-your-device/DevicePortal.md)
* Product: **windows**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-07-03:
Hi @bessobesso - maybe a silly question from me, but why do you want to change the port?

bessobesso, 2018-07-03:
I need remote Access, and default port used by webmanagement 8080 is already used by my internet service provider for router adimistration, so I would like to set the port to a different number

cyberh0me, 2018-07-04:
@bessobesso Windows Device Portal currently requires LAN connection so even if you would follow https://docs.microsoft.com/en-us/windows/iot-core/manage-your-device/deviceportal#changing-the-default-port you would have issues

bessobesso, 2018-07-04:
I did it like described in https://docs.microsoft.com/en-us/windows/iot-core/manage-your-device/deviceportal#changing-the-default-port, but this does not work in LAN, I can change the port but device portal does not respond after change, when I change back to 8080 it responds. Is there any way to get the device portal responding on other port than 8080 (in LAN) ?

cyberh0me, 2018-07-05:
if you have changed the WDP default port did you also changed the firewall settings to allow the connection ?
you could try to logon to your device with PowerShell and use [Test-NetConnection](https://docs.microsoft.com/en-us/powershell/module/nettcpip/test-netconnection?view=win10-ps) against **localhost or 127.0.0.1** to confirm WDP is listening on the new port
but because you mentioned remote access and router WDP currently is not supported in NAT environments and requires a local LAN connection between the remote management client (browser) and the device (_The Windows Device Portal (WDP) lets you configure and manage your device remotely over your_ **local network**.)

bessobesso, 2018-07-05:
Thank you very much, that helped me. I added firewall rule and WDP started to respond :)

saraclay, 2018-07-11:
Thank you @cyberh0me! Closing as this has been resolved.

closed 2018-07-11 by saraclay