MicrosoftDocs/windows-iotcore-docs #471: Inaccurate instructions

brendancmiller, 2018-07-02:
"At this point, Visual Studio will present the Remote Connections dialog". In Visual Studio 2017, this does not happen. You have to go to the project settings screen update your remote connection setting there.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b4872225-c356-8132-b15d-ea7866cc7997
* Version Independent ID: 82f7bca9-c353-5090-20c5-739833090ebb
* Content: [Deploying an App with Visual Studio - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/appdeployment)
* Content Source: [windows-iotcore/develop-your-app/AppDeployment.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/AppDeployment.md)
* Product: **windows-iot**
* GitHub Login: @bfjelds
* Microsoft Alias: **brian.fjeldstad**

bfjelds, 2018-07-02:
Thanks for the update.  Just to make sure, are you using C++?  At least previously, the dialog appeared for managed projects, but C++ projects required going to the project settings.

brendancmiller, 2018-07-02:
No. I'm using C#

cyberh0me, 2018-07-04:
are you sure you clicked the dropdown (arrow down on right side) as request ?
if i click this with vs2017 it works as described

brendancmiller, 2018-07-06:
I was just going to suggest this actually. I'm using code that I've downloaded from github. So that's likely the culprit.

closed 2018-07-06 by brendancmiller