MicrosoftDocs/windows-iotcore-docs #478: Does the latest/current build (i.e. currently 17692) include support for the RPi 3B+?

TheSilverCanuck, 2018-07-10:
Seeing "Raspberry Pi 3B+ Release" as a separate release leads one to believe that only this release supports that platform, and that the current build (i.e. currently 17692) does not.  Is this understanding correct, or is the current build supposed to also support the 3B+ model?
TheSilverCanuck@gmail.com

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: afdf6b9f-3d73-a1f2-4b57-684c07cacdab
* Version Independent ID: d9f4a048-6942-e360-933c-e4d11f4b56bb
* Content: [Release Notes for Raspberry Pi 3B+ - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/insider/rpi3bp)
* Content Source: [windows-iotcore/release-notes/Insider/RPi3BP.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/Insider/RPi3BP.md)
* Product: **windows-iot**
* GitHub Login: @zeeshanfurqan
* Microsoft Alias: **zeeshan.furqan**

saraclay, 2018-07-12:
This is correct - the Raspberry Pi 3B+ release is the only release.

closed 2018-07-12 by saraclay

egeskov, 2018-11-11:
My installation was updated through Windows Update and lost ability to connect to wired network!
I'm not sure if it possible to disable updates from happening - If not Win 10 IoT on RPi3B+ seams to be a dead fish!(