MicrosoftDocs/windows-iotcore-docs #491: On-Screen Keyboard Blocks Page Behind

scout208, 2018-07-24:
When the on-screen keyboard is open, it's not possible to interact with any elements on the page (except for the on-screen keyboard).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: afdf6b9f-3d73-a1f2-4b57-684c07cacdab
* Version Independent ID: d9f4a048-6942-e360-933c-e4d11f4b56bb
* Content: [Release Notes for Raspberry Pi 3B+ - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/insider/rpi3bp)
* Content Source: [windows-iotcore/release-notes/Insider/RPi3BP.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/Insider/RPi3BP.md)
* Product: **windows-iot**
* GitHub Login: @zeeshanfurqan
* Microsoft Alias: **zeeshan.furqan**

saraclay, 2018-08-02:
Hi @scout208 - we're still investigating this.

saraclay, 2018-08-08:
Hi @scout208 - sorry for the delay in my response. The reason for this is that we may need to release another build to fix this issue. There is still discussion internally.

scout208, 2018-08-09:
@saraclay No worries. For me, this is the only thing that is keeping from using the 3B+ for all intents and purposes. If we could just get an update on the consensus whenever it is reached internally that would be much appreciated! Thank you 👍

saraclay, 2018-08-10:
Hi @scout208 - thank you for being patient. The team was wondering if you're getting the latest Insider flights? In particular, the latest insider flight 17733 should have the OSK fix. Please let us know.

scout208, 2018-08-15:
@saraclay I'm trying it now. I will update when I have finished testing it. I thought that the latest insider flight 17733 was only for all other boards while there was a specific separate release for Raspberry Pi 3B+. Is this not correct?

saraclay, 2018-08-16:
This is true, but the OSK issue has been resolved in 17733. The release for the Raspberry Pi 3B+ is a technical preview. At this time, we recommend using the 3B or other devices with supported Intel, Qualcomm, or NXP SoCs. Do let us know if this update works for you, though.

scout208, 2018-08-16:
Okay, I think there is some confusion. I'm not having any problems with the OSK on the latest release of IoT Core for Raspberry Pi 3B. The issue I'm having with the OSK is on the technical preview. The reason I submitted this issue was to let you guys know that there is a problem with the OSK on the technical preview for 3B+ so you can know about that before you make the release.

saraclay, 2018-08-22:
Absolutely. We have definitely noted this - though if you need a fast turnaround, we do recommend the 3B until another 3B+ technical release is on its way. Closing this out for now, but feel free to re-open if anything else comes up.

closed 2018-08-22 by saraclay