MicrosoftDocs/windows-iotcore-docs #492: How to obtain the Pi 3B+ release

PeterWone, 2018-07-25:
I have an Insider account but logging in gets me 17692 which doesn't boot in my Pi 3B+

Does 17692 include everything from the Pi3B+ release or is that a branch? If it's a branch, how do I obtain and deploy it?

I know there are major changes in the B+ but it has been on the streets for more than six months. This is not helping WinIoT's image. I'm going to have to return three Pi3B+ units to my supplier if I can't resolve this soon. Lucky for me the supplier is cooperative.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: afdf6b9f-3d73-a1f2-4b57-684c07cacdab
* Version Independent ID: d9f4a048-6942-e360-933c-e4d11f4b56bb
* Content: [Release Notes for Raspberry Pi 3B+ - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/insider/rpi3bp#feedback)
* Content Source: [windows-iotcore/release-notes/Insider/RPi3BP.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/Insider/RPi3BP.md)
* Product: **windows-iot**
* GitHub Login: @zeeshanfurqan
* Microsoft Alias: **zeeshan.furqan**

scout208, 2018-07-26:
There is only one Raspberry Pi 3B+ release and that is the only release that will run on Raspberry Pi 3B+. The current build (i.e. currently 17692) does not support Raspberry Pi 3B+. 

As far as obtaining the Raspberry Pi 3 B+ release:

Go to https://www.microsoft.com/en-us/software-download/windowsiot and click the link for "RaspberryPi 3B+Technical Preview Build 17661" as follows.

[image]

After installation, you will find a file called flash.ffu under C:\Program Files (x86)\Microsoft IoT\FFU\RaspberryPi2.

I'm assuming you know how to flash that onto an SD Card using Windows 10 IoT Core Dashboard but if not I can direct you on how to do that.

PeterWone, 2018-07-28:
Thank you for being quickly helpful.

Since the team removed the B+ changes I guess it’s not simple to do a reliable universal build.

Yes, thank you, I do know how to use the FFU, I learnt that yesterday 😊

The wide variation in quality suggests that you are extremely busy responding to the changing world, so I very much appreciate your taking time to help me.

[image]<https://user-images.githubusercontent.com/9426329/43273450-08028498-90c2-11e8-91c2-3bb0e1aff4d1.png>

saraclay, 2018-08-01:
Thank you @scout208 for the answer. Closing this issue as it has been resolved.

closed 2018-08-01 by saraclay