MicrosoftDocs/windows-iotcore-docs #494: Does installing using provisioning package from WCD work if you already have an older version of the app installed on the device?

scout208, 2018-07-27:
Hi, I was just wondering if installing using provisioning package from WCD will still work even if there is already another version of the app installed on the device. I couldn't find any information about this on the docs and it doesn't seem to be working for my device.

anphel31, 2018-08-01:
Yes, this is supported. You need to ensure that both the provisioning pack version and APPX version numbers are incremented.
Here is a pointer to some provisioning information: 
https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/add-a-provisioning-package-to-an-image

anphel31, 2018-08-01:
Additional documentation: https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/appinstaller

saraclay, 2018-08-02:
Thanks for the answer, @anphel31 - closing as resolved.

closed 2018-08-02 by saraclay