MicrosoftDocs/windows-iotcore-docs #508: How to Enable GPIO Use in IoT Core on Up Squared Board

scout208, 2018-08-08:
I have IoT Core running on my Up Squared board, setup via [these instructions](https://docs.microsoft.com/en-us/windows/iot-core/tutorials/quickstarter/devicesetup#flashing-with-emmc-for-up-squared-other-intel-devices)/

However I'm unable to successfully use GPIOs. I'm running my UWP app written in C# and this is my code for opening a GPIO pin:

    gpio = GpioController.GetDefault();
    Light2 = gpio.OpenPin(GPIOPinLight2);

However, `gpio` is always null. I have this same code running perfectly fine on a Raspberry Pi 3 running Windows 10 IoT Core but it fails on Up Squared board. Is there some extra step to get the GPIO pins working?

Thank you.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6547b7e1-4e53-9510-e550-ac2d369da6d9
* Version Independent ID: a3c18768-7ee4-4e7a-2ca4-f6f6c31f342a
* Content: [Setting up your device - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/tutorials/quickstarter/devicesetup#flashing-with-emmc-for-dragonboard-410c-other-qualcomm-devices)
* Content Source: [windows-iotcore/tutorials/quickstarter/DeviceSetup.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/tutorials/quickstarter/DeviceSetup.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

scout208, 2018-08-08:
Found it. Turns out there's some things that need to be changed in the BIOS in order to use GPIOs. There was a PDF that came in the folder with the ffu in it from the Up Squared download site. In there were the instructions I needed to follow to enable use of GPIOs.

closed 2018-08-08 by scout208

kcow3, 2021-05-17:
Running into the same issue. Can you remember where you found the "PDF" mentioned above?
Or the specific settings in the BIOS you changed?

scout208, 2021-05-17:
@kcow3 I can't remember exactly of the top of my head what specific settings I changed. I would start by at least making sure you change the OS options in the bios to Windows IoT Core.

kcow3, 2021-05-18:
@scout208, Thanks! 

Changed the OS to target Windows IoT Core. And changed the following setting in the BIOS:
Navigate to "Main --> CRB Setup --> CRB Advanced --> RC ACPI Settings  --> PUP Constraints" and then enable PEP SDIO Setting.