MicrosoftDocs/windows-iotcore-docs #516: Dragonboard 410c analog audio output

endyb66, 2018-08-09:
On the Dragonboard 410c, there is a analog audio header (J7). When connecting small speakers to the direct output SPKR_L and SPKR_R, everything works. But i need to attach a 3.5mm jack, sadly on These pins is no Audio Signal and there is no audio device listed to Change to.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f3cc8f9f-ed48-fa2d-77b6-0fa8503ae30d
* Version Independent ID: e590ba03-e203-ece7-d2f5-03f22766a1f6
* Content: [April 2018 Update - Build 17134 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/commercial/april2018update)
* Content Source: [windows-iotcore/release-notes/commercial/April2018Update.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/commercial/April2018Update.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-08-15:
Hi @endyb66, I have let our team know and am trying to get you some help here.

endyb66, 2018-08-15:
@saraclay, Alright, thank you!

closed 2018-08-22 by saraclay

endyb66, 2018-09-25:
@saraclay, are there any news on this issue?