MicrosoftDocs/windows-iotcore-docs #519: When is a new release coming?

bormm, 2018-08-13:
Is there any roadmap for Windows 10 IoT on 3B+? The release is 3 month old and has a lot of issues.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: afdf6b9f-3d73-a1f2-4b57-684c07cacdab
* Version Independent ID: d9f4a048-6942-e360-933c-e4d11f4b56bb
* Content: [Release Notes for Raspberry Pi 3B+ - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/insider/rpi3bp#known-issues-in-this-build)
* Content Source: [windows-iotcore/release-notes/Insider/RPi3BP.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/Insider/RPi3BP.md)
* Product: **windows-iot**
* GitHub Login: @zeeshanfurqan
* Microsoft Alias: **zeeshan.furqan**

saraclay, 2018-08-16:
Hi @bormm, this release for the Raspberry Pi 3B+ is a technical preview. Limited validation and enablement has been completed. For a better evaluation experience and for any commercial products, please use the Raspberry Pi 3B or other devices with supported Intel, Qualcomm, or NXP SoCs. For the most up-to-date builds, we recommend using our Insider Flights.

closed 2018-08-16 by saraclay

EdiWang, 2018-09-18:
If Windows 10 is finished on Raspberry Pi 3B+, it would be ironic for me and others who posted a lot articles on blogs or hackster.io trying to prove to the world that Microsoft can be cool eveywhere, now it is just like another Windows Phone to me.

endyb66, 2018-09-18:
What's about the upcoming october update? Will the IoT Core version be released in the same timespan?

arnvanhoutte, 2018-10-30:
Microsoft, could we get some info about this? Will there be a further update with Wifi and Bluetooth support or are we stuck with this? If so, when is this planned?

iriahk89, 2018-11-07:
Hi Microsoft we recently purchase bulk of Raspberry Pi 3B+ with consideration on release version will be release soon as it already half year since the technical preview roll out. Please let us know the update on this. Thanks!

SeanGahan, 2019-02-01:
In a few months the release notes for this build will be a year old.....and still no change.  Sadly, the Raspberry PI Foundation is already talking about the RPi 4: https://www.tomshardware.com/news/raspberry-pi-4-everything-we-know,38539.html , and I'll bet the RP 4 is out before MS supports the RPi 3B+.  At this point, I do not expect MS to fix the issues with this build;  I think the lack of support is a clear indicator that MS has other interest which do not include this platform.  Makes me wonder about their commitment to IOT in general.  One thing that MS has right is Visual Studio, it rocks!

bormm, 2019-02-01:
Really sad. 6 months ago I asked this question and there is no light at the horizon. Seems my wish "Hopefully this all will not go the way of Windows Embedded, Windows CE ..." will not be fulfilled. So sad!
Do you see all these comments @saraclay ? These developers! and more are enthusiastic to use your platform and feel pure frustration. Even the last non-3b+ release is from october 2018 and there is no new insider preview at all. So is the platform dead now?
Maybe Linux + .Net Core is the best Embedded Microsoft platform then.

endyb66, 2019-02-01:
@SeanGahan, that's true, it has been a long time, since we saw the last update for Windows IoT Core, so i don't know, what is happening to this product...

PeterWone, 2019-02-10:
@bormm - Unfortunately I fear you may be right. This is exactly the same pattern I experienced with Silverlight and Windows Phone: Microsoft goes quiet on the subject and updates slow to a trickle.

That it should happen to Windows IOT is quite surprising. WinIOT is a testbed for many of the elements of the OneCore project that spawned dotnet Core, which represents the future of Windows if it is ever to leave behind Win32.  But if I understand correctly, Microsoft's revenues now come primarily from Azure and Office, and it is not beyond the realms of possibility that this is why they're so supportive of dotnet on other platforms - maybe Windows itself will be eventually be retired. It would certainly relieve them of a major cost centre -- support and maintenance of a huge and ancient mess.

@saraclay was responsible for open sourcing the Lightning drivers (memory mapped instead of message based), and there hasn't been any Microsoft input into that project for years. There have been plenty of pull requests from people fixing bugs, but not one has been merged. In particular, there's a debounce bug with a one-line fix. This is hardly a time consuming merge, and so it is very obvious that Microsoft has no interest in supporting Lightning or by implication the platform that needs it.

So, if WinIOT is an abandoned platform, what should we do? Python is not an acceptable option. However, as you mentioned, dotnet core on Raspbian is not out of the question, and it does still seem to be a supported technology.

What I don't like about both WinIOT and Raspbian is the boot time. I'm seriously considering Meadow, which runs Mono on top of μRTOS (that's mu not a U, pronounced micro-RTOS). Boot time on the order of a second. Headless, of course. The boards are reminiscent of a Pi Zero. It has the horsepower of a Pi and the analog IO and short boot time of an Arduino, and you can code for it with Visual Studio. Sitting it on μRTOS was clever because it does the heavy lifting of abstracting the platform.