MicrosoftDocs/windows-iotcore-docs #537: Windows 10 IoT Core Graphics API support

hollypang, 2018-08-27:
Hi,

I'm not sure if this is the right forum to post this question. I found the link to this github on the Win 10 IoT Core Overview page (https://docs.microsoft.com/en-us/windows/iot-core/windows-iot-core).

I would like to know if there is a public document listing the 2D/3D graphics APIs that are supported in Windows 10 IoT Core OS.  For example, is OpenGL supported?  Is DirectX supported?  I wasn't able to find this answer in the Microsoft IoT Core docs.

Thank you for your help.

-Holly

saraclay, 2018-08-29:
Hi Holly,

We do not have documentation yet on this currently, but we will look into this. Closing this out now as we'll be creating a task on our side for this.

hollypang, 2018-08-29:
Is there a ticket where I can follow up on this task?  Thanks.

saraclay, 2018-08-29:
Hi @hollypang,

I actually got back to the team, and while there is no public documentation, there’s nothing special or different about graphics with IoT Core compared to the rest of windows. If you have a DX capable driver, then the system will accelerate the work just like on a PC. If you don’t then you get software emulation like on RPi.

Does that make sense?

hollypang, 2018-08-29:
Thanks for getting back to me.  I appreciate it.  I have a few more questions.

(1) What about OpenGL?  Last I heard was OpenGL APIs are not supported on UWP apps.  
(2) I also have questions on what Medic Codecs IoT Core support (H.265, H.264, VC-1, MJPEG).  Is there a list somewhere?  If we use the same underlying graphics media hardware, is the support on media codecs the same regardless if we install Windows 10 Enterprise or Windows 10 IoT Core?  

Thanks,
Holly

saraclay, 2018-09-05:
Hi Holly,

Referring back to my note above, UWP is part of the broader Windows ecosystem - so I wouldn't treat it separately from Windows.

This is what I got after a quick search on UWP support for OpenGL.
https://stackoverflow.com/questions/39943970/uwp-support-for-opengl 

For your second question, yes, the support is the same. Closing this issue.

closed 2018-09-05 by saraclay