MicrosoftDocs/windows-iotcore-docs #608: Raspberry Pi 3 model B on Windows 10 IoT Core shutdown issue

ArAmM7, 2018-09-14:
Shutting down the device doesn't seem to do it properly, the green LED keeps blinking. A bit of googling suggested that the device doesn't shut down fully. Was just wondering if a future update will be released to fix this.

v.10.0.17134.1

saraclay, 2018-09-17:
Hi @ArAmM7 - can you be a bit more specific? How does the device not shut down fully?

ArAmM7, 2018-09-18:
With Raspbian pressing shutdown turns off the pi and the green LED turns off and doesn't blink(the red one stays on it's just a power indicator), on Windows 10 IoT pressing shutdown turns of the screen and all but the green LED blinks 7 times stops for a while then blinks 7 times again and the loop continues.

saraclay, 2018-09-19:
Thank you for the follow-up. While I am investigating this with the team, my hunch is that Raspbian and Windows 10 IoT Core behaviors sometimes are not always 1:1. While this may be expected behavior, it might also be a bug worth filing, which you can do here: https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT

saraclay, 2018-09-19:
Before you file a bug, have you tried this on flight 17758? If you haven't, give that a try.

ArAmM7, 2018-09-20:
Will be try it in the following days, thanks for the reply.

ArAmM7, 2018-09-20:
[image]
Version 17758 was not available for download, will use 17744 instead

saraclay, 2018-10-03:
Closing for now. @ArAmM7, feel free to open a new issue if the need to comes up.

closed 2018-10-03 by saraclay