MicrosoftDocs/windows-iotcore-docs #620: Audio support

jjgriff93, 2018-09-20:
There is currently only one audio device listed as supported, with no indications of good speaker and microphone arrays to use if you're developing with IoT Core. Can we add more please?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9c8da9d0-10ee-b8e6-6c11-16f1faf8a630
* Version Independent ID: e45ebd71-6334-7dd6-87a2-933210996f4c
* Content: [Hardware Compatbility List - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/hardwarecompatlist)
* Content Source: [windows-iotcore/learn-about-hardware/HardwareCompatList.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/learn-about-hardware/HardwareCompatList.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-09-24:
Hi @jjgriff93 - as it states at the top of the document, the list is not exhaustive. Our docs are open source and we encourage people to contribute content about the peripherals that are supported. With that being said, we will be more proactive about encouraging people to contribute to this document.

closed 2018-09-24 by saraclay

pbharat84, 2018-10-13:
Can you please list out some of the microphone and speaker arrays that are supported by windows IOT. I have been looking for these for a while and no luck finding them. Appreciate your help.