MicrosoftDocs/windows-iotcore-docs #639: Link to default app source

mwettste, 2018-09-25:
The link to the default iot core app leads to a repository where still the old default application is hosted. Has the source code of the new version not been published yet, or is the code hosted in a new repository and the link should be fixed?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a6854849-9812-2667-11cc-5506c8e36e22
* Version Independent ID: 94814756-3eb3-b488-778f-7ad34fd96ee6
* Content: [Windows 10 IoT Core Default App - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/iotcoredefaultapp)
* Content Source: [windows-iotcore/develop-your-app/IoTCoreDefaultApp.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/develop-your-app/IoTCoreDefaultApp.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-09-26:
Hi there - the code should be published by 10/2! We got ahead of ourselves with the documentation :)

closed 2018-09-26 by saraclay

mwettste, 2018-09-26:
Cool, thanks for the feedback :)

mwettste, 2018-10-23:
I would be interested in an update as well. The code for the sample app is still for the old version.