MicrosoftDocs/windows-iotcore-docs #664: Old Qualcomm BSP

endyb66, 2018-10-03:
Why is this new october update still using the old  2118.0.0.0 BSP for the Dragonboard 410c? I dont understand this. The current version is  2120.0.0.0, which does support much better display resolutions support and audio over HDMI.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e550244c-f4be-049b-9131-ba32bd3e9214
* Version Independent ID: 77ea5cc8-8a98-53e4-affd-a1f30538bee6
* Content: [October 2018 Update - Build 17763 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/currentcommercial)
* Content Source: [windows-iotcore/release-notes/CurrentCommercial.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/CurrentCommercial.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-10-05:
Hi @endyb66, the team is looking into this.

endyb66, 2018-10-05:
It only was a typo on the version, the newest release does contain the current BSP:

https://social.msdn.microsoft.com/Forums/en-US/dc2c5ebe-1fb4-48e4-ad47-8585566af3b8/october-update-dragonboard-410c-bsp?forum=WindowsIoT#5440791e-b3f4-405a-a430-ae6506d2d97e

saraclay, 2018-10-05:
Hi @endyb66 - yup, we confirmed this. I believe someone else has added a PR to fix this. Thanks!

closed 2018-10-05 by saraclay