MicrosoftDocs/windows-iotcore-docs #665: Dragonboard 410c boot logo orientation

endyb66, 2018-10-03:
When Setting the orientation to e.g. portrait, the boot logo (Qualcomm Snapdragon logo), does not Change ist orientation. Can this be adjusted manually or does this need a fix from on Windows IoT Core? Is it possible, to manually disable ther boot logo as long as it does not change the orientation? This would be a good Workaround for me.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e550244c-f4be-049b-9131-ba32bd3e9214
* Version Independent ID: 77ea5cc8-8a98-53e4-affd-a1f30538bee6
* Content: [October 2018 Update - Build 17763 - Windows IoT](https://docs.microsoft.com/en-us/windows/iot-core/release-notes/currentcommercial)
* Content Source: [windows-iotcore/release-notes/CurrentCommercial.md](https://github.com/MicrosoftDocs/windows-iotcore-docs/blob/master/windows-iotcore/release-notes/CurrentCommercial.md)
* Product: **windows-iot**
* GitHub Login: @saraclay
* Microsoft Alias: **saclayt**

saraclay, 2018-10-03:
Hi @endyb66,

To disable the boot U, you can go hereI: https://msdn.microsoft.com/en-us/library/windows/hardware/mt683761(v=vs.85).aspx

You won’t be able to use the Windows ICD option because that is only available for desktop provisioning, however they could manually disable the boot UI elements using bcdedit: Bcdedit -set {globalsettings} bootuxdisabled on 

Note that this bcd setting will also disable the spinning dots.

closed 2018-10-03 by saraclay

endyb66, 2018-10-12:
@saraclay, can you please reopen this issue, or can you describe what i made wrong for disabling the boot UI?